SolderedElectronics/Inkplate-10-hardware #1: Unexpected reboot by ESP32 rush current with external circuit

pasutisu, 2021-11-01:
## Environment
* Inkscape 10
* external circuit require 3.3V/<200mA power supply from Inkplate 10

## Repro Steps
1. Connect external circuit and Inkplate 10.
2. Boot Inkplate 10.
3. Execute code that booting Wi-Fi logic like `WiFi.status()`.
4. Sometimes reboot without any error message.

## Reason I thought
* ESP32 has 500mA+ rush current on booting Wi-Fi.
* This board using TPS7A26 series supply only 500mA.

## Solution I tried
* I added a 470uF capasitor to 3.3V line near ESP32, it works well without unexpected reboot.

BornaBiro, 2021-11-02:
What kind of power supply is used for the Inkplate board? Too weak power supply can cause problems, as Inkplate itself can consume 500mA while refreshing the panel and connecting to WiFi (mostly in a form of current spikes), so there is not much left for external circuits. Adding a capacitors on 3.3V rail can help with maintaining constant voltage as they can serve as a "energy storage" during current spikes.

pasutisu, 2021-11-02:
I tried to supply 5V/900mA and 5V/2A to Inkplate 10 (5V/2A was more stably a bit).

I understood that if I use my circuit with Inkplate 10 stably, I need to add a LDO or caps from your comment.
But I hope more powerful LDO to 3.3V line in next version!

Thanks for helping me :)

closed 2021-11-02 by pasutisu