SolderedElectronics/Inkplate-10-hardware #2: STL files for Inkplate 10 are not the same size.

jf1452, 2021-12-04:
I just printed the stl for the Top of the case, the cavity for the PCB is 1mm to short in the x direction, and very tight in the y direction.
I compared the stls for top and bottom and are both different dimensions. Top is 1mm shorter than bottom.
I've just scaled top to get the extra 1mm back and just started printing again.
Can you correct the STLs and even better, add the source files?

davidzovko, 2021-12-05:
Hi, that difference should be on purpose :) we have printed 100's of those and all fitted without errors. Is it possibly due to the printer tolerance? This if the first time someone reports it as well, and we have had many people print on their own as well.

I will check for the source files and add them!

jf1452, 2021-12-05:
Great, re-printed the top at 101% and got a good fit; now waiting for the bottom to finish (not rescaled); noticed it has a lip so probably going to fail.
I'm now using an Ender 5 plus, never had any scaling issues before so will recheck against the cura slice and part measurements.

davidzovko, 2021-12-06:
Let me know if that worked for you in the end!

jf1452, 2021-12-07:
Reset my printer. Full recheck squareness of printer axis; I've fixed a skew issue where the Y-Axis was not quite true to the frame; now got perfect right angles.
I've printed the STLs again, cavity for the PCB is 230.52mm x 180.42mm which looks good; your spec says PCB is 230x180 so should be a good fit. After loading the STL into FreeCAD, dimensions seem to agree.
PCB I received from Crowdsupply and soldered.com is 231.25mm x 180.46mm.
3D printer is now definitely correct; but the PCB is slightly too large.

davidzovko, 2021-12-09:
Hi, the design is already pretty "loose" so even uncalibrated printed should work. To be honest, I am not sure how can you measure to 1/10 of the millimeter using hand measurement tools? :) 

Anyway, glad to got it working for you!

jf1452, 2021-12-10:
Printer is very well calibrated now better than +/- 0.2%

300mm Verniers are quite reasonably priced, measuring to better than 0.1mm is easy enough; 300mm Vernier is a must have if you have large 3D printers. Cheap ones can be bought from your favourite  sites from €30..

Also, it helps if you work for a company which makes measuring equipment that can achieve nm of resolution and 10's nm of accuracy :).  Helps you to avoid the common measuring mistakes.