SolderedElectronics/Inkplate-10-hardware #3: Case does not fit current model

werdnum, 2023-04-01:
I had a replacement Inkplate shipped to me recently because the original one was missing a charger IC. The new one seems to be physically smaller, and lack touchpads. I don't care a great deal about the touchpads, but the new smaller design seems to not fit in the 3D printable case.

The one I received is 225x162mm total, and has 4 M3x7mm standoffs in the corners. Happily this means that I should be able to print a case on my Ender 3 once I figure out a design.

Hopefully somebody (maybe I'll get around to it) will be able to redesign the case to fit the new dimensions.

KarloLeksic, 2023-04-04:
Hello, 

We will publish the files so that everyone will have access to the cases that we print for Inkplates

werdnum, 2023-04-21:
> Hello,
> 
> We will publish the files so that everyone will have access to the cases that we print for Inkplates

Fantastic! Any updates on this?

KarloLeksic, 2023-04-25:
Sorry, this may take a while.

But here are the attached files so you can print them:
[3D printable files.zip](https://github.com/SolderedElectronics/Inkplate-10-hardware/files/11322908/3D.printable.files.zip)

rsoric, 2023-07-20:
Closing the issue as my colleague has attached the required files

closed 2023-07-20 by rsoric