SolderedElectronics/Inkplate-10-hardware #6: Reproducing assembled board requires cpl

johnholl, 2024-02-05:
Hi, I'm trying to have this board printed and assemble through JLCPBC (https://jlcpcb.com/). However, in addition to the Gerber and BOM files, they need a CPL file to determine placement of components on the board. I don't have much CAD experience, but my understanding is this can be generated from the original CAD files. Would you be able to generate and add the cpl file to this repo, or point me in the direction of how to produce my own from the existing files in the repo?

rsoric, 2024-02-12:
Hi @johnholl ,

@ifadiga , can you help out with this?

ifadiga, 2024-02-12:
Hi, 

We have continued the conversation via email, and we are resolving this matter there.

closed 2024-02-12 by ifadiga