SolderedElectronics/Inkplate-10-hardware #7: 3D Print files don't fit my inkplate 10 from soldered

raphael303, 2024-10-16:
I have an Inkplate 10 from soldered and I printed the bottom of the case here on github, this doesn't fit at all. I have ordered the inkplate 10 together with the case in black, the one who came with it works well and is overall a bit smaller than the one I just printed.

The PCB is on little stands which are bolted to the PCB, not screwed. There is no way it fits the printed one. The holes are completely misaligned. There must be some other versions around, some older 3D print files too probably.

Has anybody here any idea about different versions of the case and the PCB?

rsoric, 2024-10-16:
Hi @raphael303 ,

Could you check [this ](https://github.com/SolderedElectronics/Soldered-Inkplate-10-hardware-design)repository which contains files for the newer version of Inkplate (labeled 'Soldered' Inkplate). If your PCB is purple - it's more than likely  the 'Soldered' newer version.

Do [these ](https://github.com/SolderedElectronics/Soldered-Inkplate-10-hardware-design/tree/main/OUTPUTS/V1.3.1/3D%20printable%20files) files work better for your device?

-Rob

raphael303, 2024-10-16:
Thank you. Looks better and I am printing now. A pitty for the filament wasted. To me it was not clear there were two versions, two repositories.

closed 2024-10-17 by rsoric